polyend/TrackerBetaTesting #1332: [Firmware version 1.5.0b1] (Partly) erased samples when (re-)loading prevoius sample

Jester-Gatas, 2021-12-22:
Loving the new features of 1.5 so far!

Not sure if this is a new issue, but I couldn't find it, and didn't quite know what to search for, so here goes. 
Hit a problem with the loading space for the samples. Not really sure how to describe it as English is not my first language, but I'll try.
Lets say I have 4 samples loaded, and I for some reason want to change sample nr 2. If the new sample I choose to load is longer than the sample it replaces, the new sample eats in to the next sample's loading space. Sample nr 3 will then get a silent nothing for as long as the previous samples excess time/loading space is. Hm. Hopes that makes sense.  

* This has happened 2 or 3 times over the last 10 new songs.
* Haven't seen this in previous firmware versions. Only in 1.5.0.beta1
* It tends to happen as a project grows larger and more complex. But in each case I have been a long way from using the full sample memory.

-A

Jacknametrouble, 2021-12-23:
Please check https://github.com/polyend/TrackerBetaTesting/issues/1328

closed 2021-12-24 by ambv

Jacknametrouble, 2022-01-15:
Fixed in firmware v1.5.0