polyend/TrackerBetaTesting #1339: [Suggestion] Effect for Mutual Exclusion of Notes Between Tracks

zach-overflow, 2021-12-26:
# Effect for Mutual Exclusion of Notes Between Tracks

## Has This Been Requested?
I've spent about an hour searching for any feature requests like this, and none appear to cover what I'll be describing in this issue. However, I did find one feature request which may _possibly_ be a prerequisite for the feature I'm describing here. That possible "pre-req" feature request is described in issue #928 . It will become clearer why that issue _might_ be a pre-req for this feature as we get into more detail below. **That said, of course please let me know if I missed an issue which describes what I've written and I'll happily close this issue**!

## The Current Problem
Often, there are situations where you have two notes _in different tracks_ which you would want to be played mutually exclusive to one another, or in other words, they "choke" each other. The tracker handles a similar concept perfectly when the notes are _not_ simultaneous and in the _same_ track (For example, it's often desirable for open high-hat notes to get choked by any subsequent closed high hat note -- which works perfectly when both high-hat notes are in the same single track).

What the Tracker cannot currently handle, is if two simultaneous, or nearby, notes in two different tracks should be "mutually exclusive" and choke one or another. 

## Suggestion: New Effect for the Mutual Exclusion of Notes Between Tracks
First, let's clarify why cross-track note chokes would be a valuable feature by digging into a few concrete examples. Then we can get into some of the potential challenges associated with this feature request.

### Examples
We'll pretend the effect described in this issue is referred to as "Z" and the value refers to the track number to be mutually exclusive with (so values of "Z1" to "Z8").

#### Example 1: Probabilistically Playing One of Two Simultaneous Instruments
Below is a table representation of a scenario where you have two simultaneous notes (same step) in different tracks, and you want only one of the two notes to play at any given time.

| Track    | Note | Inst. | FX-1 | FX-2 |
| -------- | ---- | ----- | ---- | ---- |
| TRACK 1  | C3   | 01    | None | None |
| TRACK 2  | A5   | 02    | C40  | Z1   |

With the second track's note having a chance of 40%, we now have a subsequent 60% chance for the note in track 1 to play instead. The "Z1" effect ensures these two notes are mutually exclusive. This sort of setup can be extremely useful in keeping some level of interaction between probabilistic notes.

#### Example 2: Probabilistically Playing One of Two Nonconsecutive Notes for the _Same_ Instrument
The "Random Note" effect allows one to randomly play notes in a _range_, but it does not offer the ability to randomly play one of two _nonconsecutive_ notes (for example, C3 and A5). In this example, the proposed effect can do that. As with before, below is a table representation of a single step and the state of two tracks at that step. Note that this time, the instrument is the same in both tracks.

| Track    | Note | Inst. | FX-1 | FX-2 |
| -------- | ---- | ----- | ---- | ---- |
| TRACK 1  | C3   | 01    | None | None |
| TRACK 2  | A5   | 01    | C50  | Z1   |

The second track's note has a chance of 50%, and has a mutual exclusion effect for track 1, meaning track 1 will play during this step the other 50% of the time. We have now allowed for a single step to randomly play one of two nonconsecutive notes for the same instrument.

### Potential Implementation Challenges
I won't pretend to know anything about the underlying code / hardware architecture for the Tracker, so I will only talk about some conceptual challenges related to implementing this feature.

1. What if a single step has this effect in both the FX1 and FX2 slots?
	* If both FX entries point to the _same_ track number (i.e. FX1 = `Z4` and FX2 = `Z4`), then presumably one of those FX entries is simply redundant.
	* If the FX entries point to _different_ track numbers, does that imply a mutual exclusion between the other two tracks pointed to by the FX entries? 

2. From a user-perspective, it might be a bit unclear at first why a track doesn't play its note if the exclusion effect is set in a different track. For example, if we look at the 2nd example above, it is not immediately clear to a user that TRACK 1 has an implied probability for that specific step. I suppose one potential way to address this is to have some sort of subtle "highlight" color on the track's specific step(s) which are under this effect. Referring to the 2nd example again, then TRACK 1's step could have a soft highlight around it to show that it is under the exclusive effect due to another track's FX value at that step.

3. There may be some confusion if there is a "loop" in the tracks pointed to by this effect. For example, lets say that we had this scenario:

| Track    | Note | Inst. | FX-1 | FX-2 |
| -------- | ---- | ----- | ---- | ---- |
| TRACK 1  | C3   | 01    | None | Z3   |
| TRACK 2  | A5   | 02    | C40  | Z1   |
| TRACK 3  | D#4  | 03    | None | Z2   |

In this case, we have a "loop" where track 1 excludes track 3, track 3 excludes track 2, and track 2 excludes track 1 all in the same step. In this scenario, what actually will happen is 40% of the time, track 2's note will play, and the other 60% of the time, no note is played in any of the three tracks during that step. That isn't necessarily a bad thing but it may be difficult for the Tracker to efficiently the "exclusion graph" between all 8 tracks per step, and apply those exclusions properly.

4. If neither track has a "chance" note, then the mutual exclusion would always apply between two simultaneous notes. For example, lets look at this scenario: 

| Track    | Note | Inst. | FX-1 | FX-2 |
| -------- | ---- | ----- | ---- | ---- |
| TRACK 1  | C3   | 01    | None | None |
| TRACK 2  | A5   | 02    | None | Z1   |

In this example, Track 1 and Track 2 will never play their notes on this step since there is no chance effect enforcing a probabilistic note. One potential solution here is to actually build the a "probability" into this proposed effect, so the value of the exclusion effect would be something like "Z[track number-probability]", so the scenario in the third point above would have a "Z1-40" in FX-2 and "None" in FX-1.

## Conclusion
This effect would be especially useful for generative music. Particularly this effect would come in quite handy if you want to prevent the clashing of two notes / instruments, while preserving some level of randomness. This essentially would add some level of "generative awareness" between two or more tracks, which opens up a whole world of interesting probabilistic dynamics. Additionally, this feature would provide further control of random notes for a single instrument on a single step by allowing for a random choice between two non-consecutive notes (as described in "Example 2".

DRxZxL, 2021-12-28:
Gotta say i think that effect suggestion would be a huge gamechanger +10 gets my vote

Jacknametrouble, 2022-01-17:
Hey, Well I must admit this is a very professional entry. Very well explained. Thank you so much for sharing your idea in such an approachable way. The more I'm sad to say this won't have a chance to be implemented in the PT as it is. The way it's written would require rewriting some things from scratch, which just won't happen for many reasons. We'll be saving this message for future hardware iterations. Maybe we'll be able to make something like this one day. Respectfully,

closed 2022-01-17 by Jacknametrouble

zach-overflow, 2022-02-05:
@Jacknametrouble Thanks for the informative response, and for taking the time to read the proposal! I wasn't sure if this feature was possible to implement with the current hardware / OS, so I totally understand. Glad you all have this saved for any future discussions.