polyend/TrackerBetaTesting #1340: [1.4.0b1] MIDI ground loop issue

tfspsound, 2021-12-27:
Hey

Had a look but the only noise related issues seem to be Line In related, if it’s a duplicate just let me know. 
I was creating a patch on a zoia euroburo that I wanted to clock using the tracker whilst send the audio from the tracker through the zoia inputs. 
I had been using the audio inputs without issue up until the point I connected the midi cable at which point a buzzing sound was added to the signal which was only present while the sequencer on the tracker was running. 
The sound increased and decreased in pitch depending on the tempo, so I think the clock is bleeding into the audio outputs. 
The devices were connected via midi cable using the adapters provided (type B tracker, type A zoia) No other equipment was involved. The tracker was powered with the supplied power block and usb cable. 
I tested the zoia using the clock from an MPC One to confirm that the noise was produced by the tracker and I had no issue with that so it must be the tracker. 
Hope this is something you can address through software as the buzzing sound makes this setup unusable. 

Cheers

MortalJS, 2021-12-29:
Hi @tfspsound 

Can You sketch out exactly (schematically) how you have connected the devices together. 
It is mainly about the audio signal path.

tfspsound, 2022-01-03:
[image]
Hope this helps
Cheers

MortalJS, 2022-01-04:
Hi @tfspsound 

First of all, thank you for the sketch. 

I made a test on two effects I own. Unfortunately I did not get any noise when changing the tempo. 

I don't really know what could be the reason. I have encountered the groudloop effect especially with USB powered devices and generaly USB midi (for example MPC one in my case has to be powered from the strip without PE wire, then the hum disappears).

[image]
[image]

[image]

Recorded sound from the effect output.

[#1340.zip](https://github.com/polyend/TrackerBetaTesting/files/7810431/1340.zip)

tfspsound, 2022-01-05:
https://user-images.githubusercontent.com/73538421/148213699-89d8968c-0597-4926-9ced-be6098209319.MOV

Hey
Made a quick video of the issue you can hear the buzz changing pitch as I change the tempo. Using a different strip but not sure about the wiring.

MortalJS, 2022-01-06:
Hi @tfspsound 
Are you 100% sure this is not related to the zoia patch?
It sounds very similar to the effect achieved in 4:40.

https://www.youtube.com/watch?v=U2QV3T7J2Mk

Are you sure you haven't used any of the Tracker inputs/outputs in conjunction with the CV. Tracker does not have dedicated outputs and inputs and such a combination is capable of permanently damaging it.

tfspsound, 2022-01-06:
Hi

I can’t be 100% sure but I do know this issue didn’t occur when I used the same set up with my MPC One. The only difference there being that the connection to the MPC is a direct midi connection not relying on a trs adapter. 
I can’t hear anything on the clip similar to the buzzing lying under the beat on my video (which should just be a beat across 3 tracks)
If you notice on my video non of the cv inputs/outputs are patched but the midi input is connected to the trackers midi out. The audio inputs on the zoia accept both line and eurorack levels so there should be no issue there. 
Finally no I haven’t connected any of the cv outputs or inputs to the tracker. I can’t see why I would and they aren’t easy to mistake. Audio is arranged vertically on the right, CV goes horizontally across the bottom. My tracker still functions fine apart from the occasional bug.

MortalJS, 2022-01-11:
Unfortunately I am unable to help in any way at the moment. My role of testing the firmware bug has been completed. In this case I cannot consider it a software bug either and have to close this thread.

tfspsound, 2022-01-11:
You can take a horse to water…….

Jacknametrouble, 2022-01-11:
@tfspsound Thank you for your report. As this isn't anything we can confirm by reproducing. There's always a possibility of a hardware flaw. If you'll come to a conclusion that your Tracker behaves funkily, please reach out to your retailer in order to open up an RMA procedure. If you'd need further customer support from our side, please use https://help.polyend.com/submit_ticket 

Respectfully,

closed 2022-01-11 by Jacknametrouble

anemlife, 2022-02-13:
Precision: the noise is really present and disappear when unplugging the midi cable. If I do the same experiment with a TR-09+Euroburo for example: (TR09 midi out -> Zoia midi in -> Zoia output -> TR09 mix in -> TR09 output -> speakers) I don't experience any noise when a sequence is running. That means this issue is only present when using the Polyend Tracker (in my case at least)