polyend/TrackerBetaTesting #1343: Blue Light Filter / "Night Mode" /  RGB Filter / Hue degree + strength

egoshiner, 2021-12-27:
I believe a *gigantic* quality of life improvement to the Polyend Tracker would be a customizable RGB Filter setting. More applicably, a Blue Light Filter would be massively helpful. I get migraines/heavy headaches and feel much more sore in the day far earlier into the night when using screens/devices without any form of light filter or "Night Mode". 

Blue Light awakens the body aswell, making it harder to fall asleep after using the Polyend Tracker.

This could be achieved in UI in 3 popular, easy to understand and implement ways:

-As to more realistically emulate the displays' effects on the users' eyes, programs and apps like Flux and Twilight increase the strength of a Red Hue applied to the display over the course of the day. This would require the device having an idea what time it is and goes above and beyond the robustness this feature demands.

-An RGB or Hue adjustment meter would allow users to tint either light/black theme to a color of their liking, not at all unlike the level of customizability for users on the Ableton Live DAW. Just adjusting the Hue/Saturation/Brightness on their is enough to tune the software into something much less abrasive on the eyes and allow for much longer use-sessions (not that we all need them....)

-A higher-level "Night Mode"-labelled Red Hue or Blue Filter feature would be the absolute most accessible to all users. A secondary strength level can be used to adjust as to what level the feature goes. As for me, as time goes on, I find myself setting all my devices to have very aggressive filters.

Personally, this is a game-changer in terms of usability for me as I just just about always get a strong headache an hour or so into using the tracker with any strong level of focus, and I really really love the device.

I could only find Issue #1337, mentioning colorblindness. Various preset RGB filters can mitigate this for specific forms of colorblindness and these issues and much more could be tackled in a single section of the UI. For example Issue #1338 could also be addressed in such 'Display' that deserve their own lefthand section on the Config window.

bycym, 2021-12-28:
It would be awesome if we can create custom themes!

ketumatek, 2022-01-16:
> It would be awesome if we can create custom themes!

I will be a must 
+1
Love the idea

Jacknametrouble, 2022-01-16:
We are using white LEDs. This won't be anyhow doable on the current hardware. Thank you for sharing your idea.

closed 2022-01-16 by Jacknametrouble