polyend/TrackerBetaTesting #1345: An option for zero-padded row numbers

dustboxednorth, 2021-12-28:
As someone who comes from traditional PC-based trackers, I'm used to counting rows starting at `00` to, say `63`. It's a tad confusing to me when I go down to what I know as row `31` and see that it's marked as 32, even if that is more sensible in the sense of human-readability.

trutypesounds, 2021-12-28:
> As someone who comes from traditional PC-based trackers, I'm used to counting rows starting at `00` to, say, `63`. It's a tad confusing to me when I go down to what I know as row `31` and see that it's marked as 32, even if that is more sensible in the sense of human-readability.
> 
> EDIT: I am aware of the hexadecimal option, but even software trackers that have this option use padded zeroes by default.

was mentioned here #1083 and several other places, devs answer was a bit confusing and frustrating about some kind of performance issue in case implementing those

trutypesounds, 2021-12-28:
this approach blew my mind - doing a tracker without features that made trackers famous for

Jacknametrouble, 2022-01-12:
Polyend Tracker is based on a classic tracker idea, but it's not intended to be copying them 1:1. It's made to be accessible for people who had no previous experience with classic trackers.

closed 2022-01-12 by Jacknametrouble