polyend/TrackerBetaTesting #1349: Choose frequencies in the Master 5 bands EQ

moxip, 2022-01-02:
Proposal: It would be useful to allow the user to modify the 5 frequencies of the Master EQ.

Proposed use:
1- Go to the Master EQ page
2- Press one of the 5 keys below the screen + Turn the dial to choose the new frequency. 
 Releasing the button confirms the entry of the new value

Extremely useful application for polishing the master before recording.

danapeks, 2022-01-06:
Good suggestion, my immediate reaction when this was implemented was to try and change the frequencies and I was quite confused why you couldn't... Would be far more useful if you could.

wavestrike1234, 2022-02-02:
At the very least, add another 5 bands for less blunt EQ.

egoshiner, 2022-02-09:
Extremely useful