polyend/TrackerBetaTesting #1350: [v1.5.0 beta 2] Gate Length doesn’t work

gherat, 2022-01-04:
Hi, sorry you had trouble!

When you use beat slice and put 16 slices on a 1 bar loop, you should be able to put in a gate.
A gate of 16th per step plays the entire slice, a gate of 1/8th should play the whole slice plus the slice after it. a gate length of 1/4th should play 4 slices sequentially (thus a quater of the entire sample) A gate length of 1/32th should play half a slice and a gate length of 1/64th should play a quater slice.. This is how gate length work musically..

The “gate length” effect says in the manual it works in percentages. However, when using this on slices, it behaves completely different. It doesn’t make any sense. It does absolutely nothing between 7 and 100, no difference at all.. Then at <7 it just chops out random slices and behaves completely wrong. 

Please for god sakes, let slices play through and have a musically correct gate effect in musical notations tied to the bpm / beat of the tempo. 1/16th 1/8th 1/4th etc And let slices behave like a playhead startpoint instead of actually slicing it up.. Let them play through and let the user set a gatetime..

bambrose, 2022-01-04:
I agree that the gate is fundamentally flawed as it stands

MortalJS, 2022-01-06:
Hi @gherat 

Can you prepare a simplified project that illustrates the problem and attached it with the request?

Thanks, this will certainly speed up the subject.

MortalJS, 2022-01-25:
Close. No response

closed 2022-01-25 by MortalJS

gherat, 2022-01-25:
What do you mean no respons? 

__

Siebe Janssen