polyend/TrackerBetaTesting #1356: [1.5.0 b2] Un-deletable instrument after loading project from previous version..

jf-london, 2022-01-10:
With a current project left in the tracker, and the firmware is updated to v1.5.0 b2, Instrument 1 
will not be cleared (not used in any pattern) when using "Delete Unused" in the Sample Loader 
page. 

Importing another project created through updates from 1.3 through 1.4-Release-ver, will also 
see a stuck Instrument 1, unable to be cleared. 

Steps to Reproduce: 
-Boot in a 1.5.0 b2 updated Tracker. (perhaps with an already loaded project from another version) 
-Load a project built and saved from within 1.4 (Release Version) or earlier. 
-Clear all notes utilising Instrument 1.
-Go to Sample Loader page, and select Delete Unused. 
-Instrument 1, in my case, is still there. 

-This has happened in in other projects imported into 1.5.0b2. 

Other strange behaviour then manifested itself on that project like: The red select box on the 
"Instruments" column in Sample Loader, jumped into from Instrument Parameters, then created 
an elastic red box, above or below the current sample, selecting several Instruments with knob. 
This went away after a reboot. I could not re-create this one. Additionally, in the screwy project 
this appeared in (very basic too) I tried to: Sample Editor -> set sample end to 0.00, -> Crop. 
Going back to the Sample Loader screen, all Instruments were gone, except... number 1. 
This also, I could not reproduce. 

Saving this screwy project now, from 1.5.0b2, allows this to be reproduced indefinitely. No 
clearing of Instrument 1. Others will clear out as expected. 

The screwy project this is happening in, stripped down to one pattern with a few bits, and no 
use of Instrument 1 is zipped-up here. The sequence in it is mine, as is one sample; These are 
not to be considered freebies or for openly sharing, but for the purpose of demonstrating a 
software condition only, and are included for this purpose only. I do not make music for free 
or samples for public use, as a side note. Thank you. 

Oh. I don't use Windows, or Mac. Microsoft-github don't like Linux .zip or tar or .gz files it 
appears. No project. Great!

jf-london, 2022-01-11:
Not in any way intending to be obstructive, (github being Microsoft?) as a Linux-only household, this interface barfs on any dragged-in .zip or .tar or .gz (anything) files. Something to do with mime-type. I compressed the project into many formats, and nothing would upload or drag-on, hence my plea for an alternative. I try again here:

"Something went really wrong, and we can't process that file. Try again." Grrr. 
Tried again removing all full-stops in the name, just as file.GZ - and no. Incredible. 

https://github.com/github/hub/issues/1479

jf-london, 2022-01-11:
Dug up [my] upload issue - Edited my original report to include the little project archive. Thanks for your reply.

MortalJS, 2022-01-11:
Hi @jf-london 

I opened the uploaded project and, following your instructions, deleted all the notes then deleted all the instruments via the delete all unused function.

I tested this on different projects (from previous versions).
In my case, no matter if the project was previously re-saved or not the function works fine.

https://user-images.githubusercontent.com/90277871/148955482-a9c2a9c6-4d7e-408c-9171-5e398581cf82.mp4

jf-london, 2022-01-11:
Fhhh... Typical. I should have included the original in which I observed this, instead of clipping it down to 8 steps first. 
Herewith, is one that Instrument 1 will not be swept from. But please also consider - when you're working on something, you 
don't normally select every step, wipe, and repeat for every pattern; I am trying to free-up space in the middle of a project, and 
rubbish clear, then carry on (old habits from old computer days?) not delete everything first... :) 

In this attached, please look in the patterns 1 through 3, see there is NO Instrument 1 used, then try "Delete Unused" 

Thanks for trying to reproduce. Here goes... 

[test20.tar.gz](https://github.com/polyend/TrackerBetaTesting/files/7848719/test20.tar.gz)

jf-london, 2022-01-11:
Here is another. Again, I am not trying to erase everything, and then free up space - I'd just create a new project. I have added an audio clip, added a few notes to the pattern, then decided against it, cleared those notes, not the whole sequence, then gone to free up space, and Instrument 1 -the un-triggered one- is still there. In this case, clear all the notes of track 2 "kik", in every one of 6 patterns, then try to "Delete Unused" -For me, 1.5.0b2, it persists. Instrument 1 is not used anywhere else. 

Looking in the pattern file directory, it 'feels' as if there are ghost notes holding onto the stubborn Instrument, though no longer showing. EG: I delete all note events, and holding "Pattern" now shows no lit button for a now empty sequence, however the pattern files are in the directory still. (Seems removing all events to clear a pattern might be better served with a "Delete Pattern" function that clears all memory references and the pattern data file.. but then I don't know...) 

[Jason_02-v2.tar.gz](https://github.com/polyend/TrackerBetaTesting/files/7848926/Jason_02-v2.tar.gz)

Thanks again for trying these out.

jf-london, 2022-01-11:
Here is a video, as requested, showing "test20" holding onto an unused Instrument 1. 
-I have no project loaded - having selected "New Project" to clear all. 
-I load project "test20" 
-I go to "Pattern" then hold "Pattern" + ""Instrument" to show all used instruments on all 8 tracks.
-I scroll through pattern 1, up and down to show only Inst. 4, 7, and 8 are used.
-I scroll through the only two other patterns, 2, and 3, doing the same. 
-I go to "Sample Loader" and try to "Delete Unused" to hopefully clear Instrument 1.. 

https://user-images.githubusercontent.com/97456955/149003925-4d78f38d-891f-482f-a59b-c5df6a3eaafd.mp4

Jacknametrouble, 2022-01-12:
Thank you for your help and a professional approach, this should be helpful and may bring some answers. One more question, how are you powering your Tracker unit?

MortalJS, 2022-01-12:
Hi @jf-london 

Thank you for your comprehensive description of the subject.
The test20 project sufficiently illustrates the problem and will certainly make it easier to find the cause of the problem.
I am forwarding it to the developers' department.

@ziej8asek 
Use attached project
[test20.tar.gz](https://github.com/polyend/TrackerBetaTesting/files/7854823/test20.tar.gz)

jf-london, 2022-01-12:
Thanks for the compliments Jacek, Janusz. I used to work in commercial software development facing users worldwide for a well-known broadcaster in the UK... I do bang out some ugly C now and then, and commenting, and flow I certainly understand... Just getting a little older and more impatient now! ;) Happy to help. Such a cool device... Wow. That you have Mutable's (forgot her name) reverb algo in there is cool too!... I built a Shruti a while back!

jf-london, 2022-01-13:
> One more question, how are you powering your Tracker unit?

Sorry didn't answer this: (but for completeness)
I was using original adapter when this was discovered, but used a power bank (2x 18650 - ample stable current) for the later demonstration. 

I'll stop wittering on now.

Jacknametrouble, 2022-01-15:
Fixed in v1.5.0

closed 2022-01-15 by Jacknametrouble