polyend/TrackerBetaTesting #1357: [1.5.0b2] Screenshot renders incorrectly

trutypesounds, 2022-01-10:
terrible screenshot rendering, what is this...
[image]

MortalJS, 2022-01-11:
Hi @trutypesounds 

"terrible screenshot rendering, what is this..."

I would ask you to describe your problems clearly.
The way you communicated the changes in the render area does not indicate your comments. 
The interface under test is a beta version as you probably know and yes these kind of comments are needed. However, for the sake of healthy communication you could write exactly what has changed to the disadvantage and justify it. 

Thank you in advance.

trutypesounds, 2022-01-11:
@MortalJS ehem...
the screenshots rendered by device isnt looking as on the device itself (colors, layers, etc)
also not looking like the ones in the manual

Jacknametrouble, 2022-01-11:
Yes, the screenshots taken from the device level look different than the screenshots in the manual, and how it exactly looks on the device's screen (like transparency, etc). It has always been this way, and since the screenshot is an additional function it will probably stay this way. Thank you for your feedback. Best,

closed 2022-01-11 by Jacknametrouble