polyend/TrackerBetaTesting #1360: [1.5.0b2]   Inconveniently shaped TFT LCD ribbon cable inductor can *TEAR* through display connection slowly until itnon-functional days later

egoshiner, 2022-01-11:
This is absolutely a physical problem and in no way related to my testing and poking around the pre-release firmware versions. I've enjoyed testing it and providing feedback and discussing how disruptive this is to the overall  'portable DAW' market. To put it simply, I am *overjoyed* with the device.

Which is why on my second day of use, I found that pressing function keys directly under the screen started to make the display "glitch", or more accurately,

Jacknametrouble, 2022-01-11:
Hey, thank you for your message. If you need customer support in terms of hardware, please contact our team using https://help.polyend.com/submit_ticket

If there's anything suspicious going on with your hardware, you may like to reach out to your retailer for help too.
I don't think I've ever heard about the mentioned thing before.

Best,

closed 2022-01-11 by Jacknametrouble

egoshiner, 2022-01-11:
There should really be a general "tracker issues" github for information genuinely vital down the road for every Tracker everyone has as opposed to silent design changes and revisions, or even worse the lackthereof. Hardware feedback and software feedback are 2 sides of the same coin. I wouldn't emulate the Tracker on a computer and wouldn't want all of Ableton Live shoved into the Tracker. It's hand in hand