polyend/TrackerBetaTesting #1364: [Firmware 1.5.0 Official] Bug Performance mode

Jeymadcat, 2022-01-15:
## Hi, sorry you had trouble!

When i have only 4 tracks in playing... if i select another track empty and i play step fx (roll-step-reverse) with pad ! i have bug with long noise come inaudible ...
but i dont have this bug with beta2

trutypesounds, 2022-01-15:
is it the same as in #1111   ?

Jacknametrouble, 2022-01-15:
Can we please get a video and the project? Would be very helpful. Thank you.

Jeymadcat, 2022-01-16:
the bugs reach significantly on pti instruments but apparently also on version 1.5.0b2
I wanted to make a video yesterday on the two versions on my track where the bug came from but strangely it worked, I don't have a pti in this track.
but yes it looks like the bug is identical to #1111
I still made videos, and I'll tell you more if the big bug comes back in a track with no pti: ) sorry for my google English! : )

Jeymadcat, 2022-01-16:
as I reloaded the firmware several times this bug #1111 did not come back directly but it could come back if I reload the firmware 1.5.0>>1.5.0b2>>>1.5.0....etc

Jacknametrouble, 2022-01-16:
I'm unable to replicate this. If you can't catch this on a video, then maybe you could provide a project where this happens?
One more question. How are you powering your Tracker?

Jeymadcat, 2022-01-16:
Original Polyend Power usb adaptater

trutypesounds, 2022-01-16:
> I'm unable to replicate this. If you can't catch this on a video, then maybe you could provide a project where this happens? One more question. How are you powering your Tracker?

PLENTY OF VIDEOS HERE #1111

Jacknametrouble, 2022-01-16:
@trutypesounds I'm unable to replicate and got no confirmation on #1111 so far. @Jeymadcat please try using another stable powering source and see if this change anything. Maybe the original AC adapter is causing issues?

Jeymadcat, 2022-01-16:
after test.... if u take a analog kick from polyend instrument pack(pti) put on 4x4/32 and u play it in perf mode and "sample playback" put on reverse....the sound is cut....no sound....the same bug from 1.5.0 and 1.2.0b2

Jeymadcat, 2022-01-16:
no problem with sample kick from wav

Jacknametrouble, 2022-01-16:
@MortalJS please check, maybe you'll have more luck catching this.

trutypesounds, 2022-01-16:
@Jacknametrouble will check this out in 1.5.0 tomorrow

trutypesounds, 2022-01-18:
> 

no problem here with that

trutypesounds, 2022-01-18:
> @Jacknametrouble seems like re-saving those projects affected in #1111 solving this issue, i'll report another one in case if its happen again, i really hope that it wouldn't
> 

https://user-images.githubusercontent.com/17484006/149969661-6e230130-da34-4d69-a0f2-6e79604a5f72.MOV

> p.s. please look into this suggestion #1173 it could keep the jog and the func buttons under the screen to live more )

@Jacknametrouble damn, its still here, just did a fresh test project from scratch - step repeater causes a glitch on empty channels, check the video and the project attached
[glitchkurwa.zip](https://github.com/polyend/TrackerBetaTesting/files/7890321/glitchkurwa.zip)

trutypesounds, 2022-01-18:
well, i just another random project from scratch just to be shure, at first its behave the same - step repeater causes some sample retrig with short loop leakage on empty channel, then i saved the project, opened another then opened this new one again - the glitch leak its still there.... check the vids

https://user-images.githubusercontent.com/17484006/149974613-1e4a33b8-c7e3-4ddd-9027-37618b081425.MOV

trutypesounds, 2022-01-18:
https://user-images.githubusercontent.com/17484006/149974681-bbb7f7e0-73a9-4deb-a162-8ab2496fd90a.MOV

trutypesounds, 2022-01-18:
nah, glitch still here, tried re-savings, reboots, its very obvious if u add patter play mode set to C4 and it starting to leak to all channels and is very loud

https://user-images.githubusercontent.com/17484006/149982402-702150a4-0644-4762-ba53-3f942a2f8fab.MOV

Jacknametrouble, 2022-01-18:
Guys, we really appreciate your enthusiasm. But please let's keep this place tidy and let at least be polite.

Jacknametrouble, 2022-01-18:
@Jeymadcat how are you powering your Tracker unit?

trutypesounds, 2022-01-19:
> @Jeymadcat how are you powering your Tracker unit?

polyend cable and adapter

MortalJS, 2022-01-19:
Hi @Jeymadcat and @trutypesounds 

I have checked the reported problem. I qualify it as a bug.

Way to reproduce.
- Project file 
[test 1364.zip](https://github.com/polyend/TrackerBetaTesting/files/7897633/test.1364.zip)
- Performance mode
- Activate empty tracks (4,5,6,7,8)

The effect occurs when two effects are combined ( the problem is most probably the "step repeater")

Therefore, I am leaving this thread open and closing all duplicates.

trutypesounds, 2022-01-19:
> Hi @Jeymadcat and @trutypesounds
> 
> I have checked the reported problem. I qualify it as a bug.
> 
> > @ziej8asek
> 
> Way to reproduce.
> 
>     * Project file
>       [test 1364.zip](https://github.com/polyend/TrackerBetaTesting/files/7897633/test.1364.zip)
> 
>     * Performance mode
> 
>     * Activate empty tracks (4,5,6,7,8)
> 
> 
> The effect occurs when two effects are combined ( the problem is most probably the "step repeater")
> 
> Therefore, I am leaving this thread open and closing all duplicates.

Thank you!

closed 2022-03-19 by MortalJS

Jacknametrouble, 2022-04-29:
Fixed in https://github.com/polyend/TrackerBetaTesting/releases/tag/1.6.0b2