polyend/TrackerBetaTesting #1365: Assignable Track Outputs via Stereo and Headphone Jacks

LYSENKOCORP, 2022-01-16:
## Hi, we're excited to hear your idea!

Before you start, did you check whether your feature was already requested?
Searching might save you quite a bit of time here.

Did a search of the feature requests and didn't see this idea suggested yet. 

If tracks could be assigned to independent outputs it would be huge bonus on top of how i am using the tracker my dub style electronic set.

It could be implement like the erica synths x sonic potions LXR-02 which has two stereo outputs and two assignable outputs and any of the voices can be assigned to any of the 4 outputs. 

However the implementation on the tracker could utilize the headphone output along with the stereo output. 
So for example track 1 & 2 could be sent to Stereo L track, 3 & 4 to Stereo R, 5 & 6 to Headphone L, and 7 & 8 to Headphone R. Or any other permutation you could think of.

The signals could be split using the dongle that takes stereo 1/8th " input jack to two mono 1/4" output jacks

This would allow me to separate a few channels and effect them with my mixer send/return independently of other tracks in the tracker i would prefer to keep dry. 

Thanks for such an excellent product all ready and for giving the community space to contribute ideas!

Jacknametrouble, 2022-01-16:
Thank you for your feedback. Such functionality won't be possible in the current hardware.

closed 2022-01-16 by Jacknametrouble

LYSENKOCORP, 2022-01-16:
I know haha i came here to delete before anyone saw it. I forgot it doesn't have separate stereo and headphone outs!

So embarrassed

Jacknametrouble, 2022-01-16:
No worries.