polyend/TrackerBetaTesting #1366: [1.5.0 final] Un-selectable instrument parameters suddenly

jf-london, 2022-01-16:
Bug title: Selecting parameters in Instrument Params don't select, entered from playing project. 

+I cannot reproduce this+ But am posting it in case anyone else stumbles upon it. A video is included. 

-I updated to v 1.5 final on 16 January, 2022. 
-I reset settings in the firmware update area of config afterwards. 
-I "Create New Project," then loaded in one saved from 1.5 beta 2. 
-I pull in a "sample" file from a previous version project directory, trim it up, and "put him on the grid" as you do. 

It was in "Play" and I went to instrument settings and with a particular instrument selected, went to the filter to 
adjust an an already applied "Low Pass" filter cutoff... This is when it all started, doctor. 

It suddenly wouldn't 'take' the selection of any parameter on the "Instrument Settings" main screen. Any tap on a 
function key (below) to select a parameter, such as "Filter Type" or "Cutoff" would only be outlined in red so long 
as I held the button down. The knob/wheel would not change anything. Entering "Effects," none of the 4 parameters 
would gain a red outline box at all upon touching their function key. I could return to the main "Instrument Settings" 
screen, and the same as before - red box only lit when I hold a parameter button down. 

I left the function by pressing "Pattern" and then stop and went on as usual. 

I have made a video of this, included here. 

The project was made from the test20 one, located in issue #1356 (Instruments not being cleared) 

https://user-images.githubusercontent.com/97456955/149681140-73d26aaa-0f9f-48d8-97ff-ef2b031f03a9.mp4

MortalJS, 2022-01-17:
Hi @jf-london,
Thanks for your feedback.

Unfortunately, I have not been able to reproduce this either. 

If it is not confirmed in the near future. I am closing the thread.

trutypesounds, 2022-01-18:
> Bug title: Selecting parameters in Instrument Params don't select, entered from playing project.
> 
> +I cannot reproduce this+ But am posting it in case anyone else stumbles upon it. A video is included.
> 
> -I updated to v 1.5 final on 16 January, 2022. -I reset settings in the firmware update area of config afterwards. -I "Create New Project," then loaded in one saved from 1.5 beta 2. -I pull in a "sample" file from a previous version project directory, trim it up, and "put him on the grid" as you do.
> 
> It was in "Play" and I went to instrument settings and with a particular instrument selected, went to the filter to adjust an an already applied "Low Pass" filter cutoff... This is when it all started, doctor.
> 
> It suddenly wouldn't 'take' the selection of any parameter on the "Instrument Settings" main screen. Any tap on a function key (below) to select a parameter, such as "Filter Type" or "Cutoff" would only be outlined in red so long as I held the button down. The knob/wheel would not change anything. Entering "Effects," none of the 4 parameters would gain a red outline box at all upon touching their function key. I could return to the main "Instrument Settings" screen, and the same as before - red box only lit when I hold a parameter button down.
> 
> I left the function by pressing "Pattern" and then stop and went on as usual.
> 
> I have made a video of this, included here.
> 
> The project was made from the test20 one, located in issue #1356 (Instruments not being cleared)
> 16Jan2022_VID_20-22_noselect.mp4

tried exactly as u said, no luck to reproduce
can you reproduce it once again yourself? if yes, try this once again with another power adapter, or power your tracker via usb somehow. those polyend adapters and cables supplied with the tracker can cause instable behaviors.

Jacknametrouble, 2022-01-18:
Yes, we had some cases where the original AC/cable was causing strange behaviors.

jf-london, 2022-02-03:
Thank you Jackname' and truetype' for the suggestion - but I rarely use the original adaptor; once to see it tirn on after purchase, and recently to disprove it being the cause in the last bug I reported (#1356). I use a beefy 2A 2x18650-based power pack bought specifically for the unit. This is just a bug in a jump operation somewhere that doesn't call some latch code or something? trying to find it without an exact 'steps to follow' (to go through the source accurately) wouldn't be easy. (I have written and modified .C for years for various things.) This didn't cause me to lose any audio or project work, or crash, so wouldn't be hign on a "how to spend hours of your life" task list! ;)

MortalJS, 2022-07-14:
Close. Cannot reproduce.

closed 2022-07-14 by MortalJS