polyend/TrackerBetaTesting #1367: [Official 1.5.0] Annoying crashes

anemlife, 2022-01-16:
Just updated to 1.5.0 Friday and already had about 5 crashes while using the PT this weekend. Had no problem for weeks with the 1.5.x beta versions using the exact same setup. 

This happens completely randomly when playing patterns with multiple projects and is getting pretty annoying as I have to unplug the device + I sometime lose a few patterns after restarting it.

Jacknametrouble, 2022-01-16:
How are you powering your Tracker unit?

anemlife, 2022-01-17:
Official USB cable, sometimes with the included power brick, other times taking power from my PC Tower. Crashes happened in both cases.

anemlife, 2022-01-17:
I will give it a try tonight. If it happens again with another cable I'll send the project I was working on when the crash occurred.

colinmcardell, 2022-01-20:
I'm also experiencing consistent crashing/hanging since updating to firmware 1.5. I'm not sure if it is the same problem. I seem to have narrowed down steps to reproduce.

I've tried the below steps using two different USB cables to power the unit, powered via USB port on a MacBook Pro, and powered via a 5.25v 2A USB PSU. No difference, it still crashes/hangs.

Steps to reproduce:
1. Create a new project
1. Load up the "Synth shots.pti" instrument from the new Polyend instrument pack.
1. "Perform" a bunch of button mashing and sliding around on the grid (playing notes/slices at random really quickly, and on top of each other)
1. Eventually the tracker will hang with the buttons most recently pressed illuminated
1. To get the tracker back to a usable state it requires power cycling by way of unplugging the power cable