polyend/TrackerBetaTesting #1368: [1.5.0] Render selection freezes PT when using external clock sync

rbgazola, 2022-01-17:
Hi Polyend Team,

Everytime that I try to render a selection from the pattern view, and the clock is being sent to the PT from a external instrument, I got the PT in a freeze state and it don't come back until a hard reset. The render selection function works as usual when in Internal clock mode.

MortalJS, 2022-01-17:
Hi @rbgazola 

Can you upload simple project file. Unfortunetly in my device i dont have this problem.

Also try swapping the power supply and cable to be sure. In general, it can be a phone power adapter (the simpler the better) and a USB-C cable.

MortalJS, 2022-01-18:
Hi @rbgazola 

There is a link to manual.
https://polyend.com/manuals/tracker/#audio-rendering

"The MIDI Clock has to be set to Internal in order to render the stems/master tracks."

closed 2022-01-18 by MortalJS