polyend/TrackerBetaTesting #1370: [Firmware version] Bug title

trutypesounds, 2022-01-18:
## Hi, sorry you had trouble!

### Steps to reproduce

Please describe the steps to reproduce the bug. Tell us what you noticed
happening and what you expected to happen instead.

### Attachments and links

Often the best way to show the problem is to record a short video,
attach the Tracker project you were working on, and provide links.

Pro-tip: You can drag and drop images and `.zip` files directly to the edit
box when writing your bug reports. Videos are too big as attachments, you
will have to use an external service for this and provide us links. Please
don't use services that delete files after short periods of time.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")

* Does this bug happen in previous firmware versions?

* Does this bug happen in every project?

Jacknametrouble, 2022-01-18:
Thank you for your suggestion. This is a workflow type, not a bug.

trutypesounds, 2022-01-18:
> Thank you for your suggestion. This is a workflow type, not a bug.

kinda workflow thing, but project are not loading in the end when using open button with all those save dialogs

Jacknametrouble, 2022-01-18:
Yeah, I took a second and third look, and you got a point. Please provide the problematic project. Does this happen on these particular projects only or on the others too? And a classic question - how is your Tracker powered?

trutypesounds, 2022-01-19:
> Yeah, I took a second and third look, and you got a point. Please provide the problematic project. Does this happen on these particular projects only or on the others too? And a classic question - how is your Tracker powered?

it happens with any projects (clearly visible in the video attached, tried to open any project it goes into many save dialogs and then nothing happens), tracker is powered via polyend cable & adapter

MortalJS, 2022-01-19:
Hi @trutypesounds 

I approve this problem as a bug.

Method of restoration:
As in the video

>@ziej8asek

closed 2022-03-19 by MortalJS

Jacknametrouble, 2022-04-29:
Fixed in https://github.com/polyend/TrackerBetaTesting/releases/tag/1.6.0b2