polyend/TrackerBetaTesting #1375: [v1.5.0] Clock sync is messed up if clock sync delay is set to -6.5

martinruyant, 2022-01-21:
## Hi, sorry you had trouble!

### Steps to reproduce

- created a simple click track on my deluge, panned it right
- did the same on the tracker, panned it left
- connected the deluge to the tracker via usb
- on the tracker, set clock in to usb, clock out to off, transport in to usb, transport out to off, midi out to off

At this point I tried 3 different things:
- set clock sync delay option on tracker to -6. Tracker is slightly late
- set clock sync delay to -7. Deluge is slightly late
- set clock sync delay to -6.5. Over just a few seconds the sync between the two becomes completely wrong. Seconds worth of lag. Almost like dj'ing two tracks where one plays at 120bpm and the other at 125bpm. The sync issue would get bigger and bigger.

There may other problematic values, but "-6.5" is definitely one of them. Reproduced this many, many times. Even rebooted the tracker - issue remains. It's too bad because -6.5 seems to be exactly the value I need to set to have the deluge and tracker in perfect sync.

Video illustrating the issue:
https://www.youtube.com/watch?v=2Lx3Uu3OmUM

* How often does this bug happen? (every time, often, "it only happened once")
All the time

* Does this bug happen in previous firmware versions?
* I don't know, I recently bought the tracker.

* Does this bug happen in every project?
* Have not tested this. Spending a couple of hours going mad was enough for me :)

Jacknametrouble, 2022-01-21:
Hey, Thank you for your feedback. Unfortunately, we do not have the Deluge to our disposition. And all the other we have in the office (there's plenty) is not really allowing us to reproduce this issue. Few additional questions:

- does this also happen if you slave Deluge off of Tracker clock?
- when you start playback on Deluge, can you show us the Song screen on Tracker to make sure it's tempo-synced?
- when there's playback from both devices, can you change the tempo on Deluge up and down to see if Tracker also speeds up and slows down? (it should).

Looking forward to hearing from you.

martinruyant, 2022-01-21:
Hey Jacek,

The issue does not happen if Deluge is configured as slave.
Here is a video showing what you asked.  As you can see, tempos are synced:
https://www.youtube.com/shorts/sgK9zn2hQ5c

MortalJS, 2022-03-19:
#1136 Duplicate

closed 2022-03-19 by MortalJS