polyend/TrackerBetaTesting #1376: Sample Recorder: Play pattern automatically when hitting Record / Shift-Record?

bahutu, 2022-01-23:
Dear Developers <3
**Would it be possible to let a pattern start playing automatically when hitting Record in the Sample Recorder? For example by pressing Shift + Record or so?**

Use case: I program a MIDI pattern inside the Tracker. But instead of letting my synth (or any other external gear) play the pattern, I'd like to capture the audio as a WAV file and play it back (in sync) as a sample, instead of the external synth. That way, I could erase the MIDI events and condense multiple MIDI tracks into one single audio track. Furthermore, some synths can only play one patch at a time, so to use both a bass and lead patch, I have to capture at least one of them as audio. Last but not least, that way I could take the Tracker with me without carrying a synth around as well.

It is possible to hit both Record and Play at the same time (which is wonderful), nevertheless this introduces a slight inaccuracy due to the user (me) not pressing both keys at exactly the same time. Thus it would be really nice if the Tracker would take care of that via a "Record + Play Pattern" functionality.

This is an optional feature request. So don't feel obliged to implement it. Yet I guess, the use case might be quite wide-spread and thus appreciated by many.
(The Tracker - as it is - is absolutely amazing! Thanks for your work, dedication, firmware updates and bug fixes!)

reubenfinger, 2022-01-24:
A "threshold record" function would also work for this purpose, ie the tracker records when a certain volume set by the user is reached, would also be a welcome enhancement.

joeyfivecents, 2022-01-28:
I would also like to request this. It would help if you could hear the other tracks playing the pattern while you record the audio from midi instrument.

bahutu, 2022-01-31:
> A "threshold record" function would also work for this purpose
Yes and no. Let's say the first MIDI note starts playing on step 5, then (afterwards) the recorded audio sample would have to be put manually on step 5 as well; whereas if the pattern playback starts in sync with pressing 'record', one could always put the recorded audio on step 1 and it would be in sync automatically.
So... well, both of those things would do the trick, indeed.

Regarding joeyfivecents: I guess there might be hardware restrictions not allowing the device to play back samples and to record audio at the same time. Although it would be nice as well, I think it is slightly off-topic with regards to my suggested enhancement, since it's about sample playback, not MIDI playback. Never mind.