polyend/TrackerBetaTesting #1378: [1.5.0] Crash on playback of demo song after export to .it ->import cycle

r4dian, 2022-01-24:
### Steps to reproduce

- Using firmware 105.0
- Export the song "Sam Slater - To Know a Vale", which was included with the tracker, as an .it
- Import the mod
- Go to the song screen
- Play the project

### Attachments and links

( file attatched )
[Sam Slater - To Know a Vale.zip](https://github.com/polyend/TrackerBetaTesting/files/7928996/Sam.Slater.-.To.Know.a.Vale.zip)

### Other important information

Happens every time on the Tracker.

OpenMPT loads it - though it plays back silently - none of the samples seem to have audio data when viewed in Modplug's sample editor.

MortalJS, 2022-01-25:
Hey, It’s good to hear from you. Thank you for your feedback.

I don’t really think that accomplishing full compatibility in terms of exports (and imports) would be ever possible. Even with considering a standardised level of compatibility, we won’t ever be able to convert or provide data to convert our projects one to one. There’s a variety of tracker applications out there, working on a different basis and that’s also why we have to limit the import/export capabilities to the currently supported parameters and values.

The same applies to the other threads relating to .it

closed 2022-01-25 by MortalJS

r4dian, 2022-01-25:
I don't expect "full compatibility", but I think being able to open and play a file the device itself saved should be considered bare minimum functionality.