polyend/TrackerBetaTesting #1379: [1.5.0] Audio output stops seemingly at random

r4dian, 2022-01-24:
### Steps to reproduce

Sometimes in use the audio output randomly stops. 
There's not a reliable set of steps to reproduce this, but its happening very often, and in the following screens:

- sample recorder
- sample editor
- pattern editor

### Other important information

* The meters, playhead cursor etc still move as if it it were playing normally
* Happens about twice an hour
* Never happened before (I upgraded to 1.5.0 today)
* Heaphones tested with another device to confirm the error lies in the Tracker
* Happens in any project
* A reboot is not enough to resolve the issue once it occurs, only powering off and disconnecting the power before restarting cures it.

Jacknametrouble, 2022-01-26:
What powering source are you using? If it's happening on the original AC adapter/cable, please try using an alternative source and see if it changes anything.

r4dian, 2022-01-26:
I've used the included plug & PC USB, but both with the included cable. 
I'll test a different cable...

Jacknametrouble, 2022-01-26:
Some of the original AC units were causing funky behavior. USB port can also be not so stable. Besides the cable, you may like to try another USB adapter, like your phone's one for eg.

r4dian, 2022-01-27:
Yes, it was resolved by using phones plug instead.
Is there any way it you could add a warning (at start up, or while running) if the power supply is borderline?

If not, just close this, thank you.

Jacknametrouble, 2022-01-28:
Thank you for the update. Glad you've figured it out. Thing is, it's not a very common issue. Seems like the manufacturer and supplier of the AC adapters we use, hooked us up with a number of problematic units in one of the batches. But that's why we always ask customers about it when the Tracker is behaving irrationally.

closed 2022-01-28 by Jacknametrouble