polyend/TrackerBetaTesting #1380: [1.5.0] Samples from a "sample mode" .it modules are silent once imported & have 0 length

r4dian, 2022-01-24:
### Steps to reproduce

- Create a new .it format song in OpenMPT
- Create a sample by drawing the waveform (example 1, attached)
- OR - import a sample (example 2, attached)
- Add a single note playing the sample in the first (and only) pattern
- Save the song in either normal or "compatible" .it format.
- Import on the Tracker.
- Try to play the song
- Go to sample editor & check the end & loop end

### Attachments and links

Example one: hand drawn wave sample 
[Mod.zip](https://github.com/polyend/TrackerBetaTesting/files/7929190/Mod.zip)

Example 2: imported 44k1Hz sample
[Mod2.zip](https://github.com/polyend/TrackerBetaTesting/files/7929191/Mod2.zip)

https://youtu.be/rb3LPCnhQto

MortalJS, 2022-01-25:
Hey, It’s good to hear from you. Thank you for your feedback.

I don’t really think that accomplishing full compatibility in terms of exports (and imports) would be ever possible. Even with considering a standardised level of compatibility, we won’t ever be able to convert or provide data to convert our projects one to one. There’s a variety of tracker applications out there, working on a different basis and that’s also why we have to limit the import/export capabilities to the currently supported parameters and values.

The same applies to the other threads relating to .it

closed 2022-01-25 by MortalJS

r4dian, 2022-01-25:
Hello,

I don't expect full compatibility, but do expect the basics like "samples actually play audio" to work, especially when the products own website claims

`Compatibility with MOD files allows you to import and finish tracks from back in the days or export and finalize them in the modern software trackers.`

If by "we have to limit the import/export capabilities to the currently supported parameters and values." you mean that you don't intend any further development on this feature, then I feel I have been missold. 

In the import feature's current state, "tracks from back in the days" either don't play at all, or are unlistenable garbage (my younger-self's talent non-withstanding 😉)