polyend/TrackerBetaTesting #1381: [1.5.0] one-shot samples from "instrument mode" .it files are looped when they should not be.

r4dian, 2022-01-24:
### Steps to reproduce

- Create a new .it format song in OpenMPT
- Import a sample
- Add a single note playing the sample in the first (and only) pattern
- Go to the instruments tab & click "new instrument"
- Accept when it offers to convert existing samples into instruments first
- Save the song in either normal or "compatible" .it format.
- Import on the Tracker.
- Try to play the song
- Go to sample editor & check the playback mode

### Attachments and links

[Mod3.zip](https://github.com/polyend/TrackerBetaTesting/files/7929331/Mod3.zip)

[image]

[image]

MortalJS, 2022-01-25:
Hey, It’s good to hear from you. Thank you for your feedback. 

I don’t really think that accomplishing full compatibility in terms of exports (and imports) would be ever possible. Even with considering a standardised level of compatibility, we won’t ever be able to convert or provide data to convert our projects one to one. There’s a variety of tracker applications out there, working on a different basis and that’s also why we have to limit the import/export capabilities to the currently supported parameters and values.

The same applies to the other threads relating to .it

closed 2022-01-25 by MortalJS

r4dian, 2022-01-25:
Again, I don't expect "full compatibility" as the capabilities of the Polyend Tracker, and Impulse Tracker don't match 1:1, but a basic song with one-shot samples and no effects used should at least sound the same in both, otherwise why even have this import feature at all ?