polyend/TrackerBetaTesting #1385: Performance Mode - switch from loop mode to song mode mutes effects

PITmast, 2022-01-28:
Hello everyone
Let me mark one issue in Performance mode. I guess it's a bug.

1. Start song in loop mode.
2. In Performance mode mark a track and turn on any punch-in effect. Make sure you can hear a difference.
3. Switch smoothly to Song mode by Shift+Play.
4. Track is still marked and effect is on but you can't hear it!
5. Nothing happens when you on/off punch-in effects!
6. The only way is to re-mark a track!

Regards
Piotr

Jacknametrouble, 2022-01-31:
Thank you for your feedback. Any chance for a video illustrating the issue + the project folder where it happens? Thank you,

PITmast, 2022-02-02:
Hi

It doesn't depend on project. Create new one, load one sample, create two patterns.
https://mega.nz/file/DvQTyKqb#2_HZ1WM_gUhuPqlfD6Cdvfux3x_ALF0CaTbRkq6qIoQ

best regards
Piotr

MortalJS, 2022-02-06:
Bug confirmed.
@ziej8asek

closed 2022-03-19 by MortalJS