polyend/TrackerBetaTesting #1387: Sample Loader - Colour code Instruments using Note, Instrument, Fx1, Fx2 keys

Aisjam, 2022-01-28:
## Idea
In the sample loader page, I would like to use the colours from the Note, Instrument, Fx1, Fx2 keys to highlight individual loaded instruments. 
- pressing the a coloured key changes the selected instrument to that key's colour
- pressing a different key changes the selected instrument to that key's colour
- pressing the same coloured key makes the instrument default colour again.

## Why
I use the rows from the grid to organise my instruments into percussive, leads n bass, pads and texture. however if i make a song where i manually manipulate an instrument in performance mode. It takes a bit to get into that instrument. This would be a useful visual cue to organise instrument types and/or also highlight specific instruments so they are easily identifiable.

colonel-blimp, 2022-01-31:
I'm really digging these ideas using the coloured keys to assign colours  

The @Aisjam colour keys trilogy (so far):
1. #1186
2. #1215
3. #1387

Aisjam, 2022-01-31:
> I'm really digging these ideas using the coloured keys to assign colours
> 
> The @Aisjam colour keys trilogy (so far):
> 
> 1. [Song Mode : Colour code song blocks using Note, Instrument, Fx1, Fx2 keys #1186](https://github.com/polyend/TrackerBetaTesting/issues/1186)
> 2. [Track Mixer : Colour tracks in Edit mode by Note, Instrument, Fx1, Fx2 keys #1215](https://github.com/polyend/TrackerBetaTesting/issues/1215)
> 3. [Sample Loader - Colour code Instruments using Note, Instrument, Fx1, Fx2 keys #1387](https://github.com/polyend/TrackerBetaTesting/issues/1387)

I feel this could be a really useful UI improvement adding context sensitive functions to the different pages. Every now and again I see a different potential for these buttons.