polyend/TrackerBetaTesting #1390: [1.5] No OFF/CUT/FADE on fill note

Raadiolained, 2022-01-29:
### Steps to reproduce

Set fill. Set note, steps to 4. Set some random notes and fill. 
Now I want to add OFF to these notes, so I move 2 steps below and start fill from there, but there is no OFF/CUT/FADE in fill section.

### Attachments and links

No

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
Yes

* Does this bug happen in previous firmware versions?
 Yes

* Does this bug happen in every project?
Yes

Jacknametrouble, 2022-01-29:
Thank you for your feedback. A lack of functionality is not really a bug. Try duplicating the existing pattern using copy-paste and then mark all the steps and use the jog to turn them all into OF/CUT/FADE to your liking. Best,

closed 2022-01-29 by Jacknametrouble

Raadiolained, 2022-02-18:
I did this, marked all the notes, but it did not work.

Secondly, I can not mark individual notes, only whole sections, so the OFF message would overwrite note messages even if the above mentioned suggestion would work. Am I misunderstanding something here?