polyend/TrackerBetaTesting #1391: [1.5.0] Preview does not play the whole slice in pattern mode

Raadiolained, 2022-01-29:
Beat slices do not play until the end when previewing.
This is major issue because you can have two slices with same beginning but different sounding end. I have double kick in my slice, but I hear only the first kick in preview, leaving me the impression I insert only 1 kick. When I play the pattern, I hear 2.

### Steps to reproduce

1. Slice up a sample/beat. (24 sequal slices in my case)
2. place one on the pattern
3. play the pattern
4. stop the pattern and preview the slice using preview step button and/or playing the slice from the LED grid
5. notice that previewing the slice cuts out about half-way

### Other important information

* How often does this bug happen? Every time.

* Does this bug happen in previous firmware versions? I don't know.

* Does this bug happen in every project? Yes.

Jacknametrouble, 2022-01-29:
I was trying to replicate the mentioned thing, without success. Can you please provide a video showing it? Thank you for your help!

Raadiolained, 2022-01-30:
I captured it on video. Sorry if audio and video is not 100% in sync, I had to use separate devices for audio & video.
Instrument file link is in the video description.
Here is the video (unlisted): https://youtu.be/9Ng6LxwpuNk

Jacknametrouble, 2022-01-30:
Thank you for the video @Raadiolained

Raadiolained, 2022-01-30:
I noticed same behaviour in another project, but this time envelope was ON. It was long 5 second sample set to loop. 
Preview only plays short time: in my case nothing was heard since sample was with such a long fade in (sample itself, not envelope). 

Expected behaviour would be: If I hold the preview button down, note plays and loops (if set to loop) as long as button is down.

MortalJS, 2022-01-31:
Hi @Raadiolained 

Can you attach a simplified project that shows this event. In detail with the sample that was used in the beat slice.

This will certainly speed up the verification process. 
Thanks

Raadiolained, 2022-01-31:
Hi.

Here is the simplified project:
https://easyupload.io/97dj5l
Link expires in 30 days.

I deleted everything else and left only samples where issue is present the most.
Check first 2 patterns, play the pattern, and then preview step, and also preview using grid/pads. It acts as in the video.
I also noticed it persists when I preview the same samples (already loaded) in sample loader view.

MortalJS, 2022-02-01:
Hi @Raadiolained 

Thanks for your simplified project. 
I understand what the case you describe is about.
I need to consult it with the programmers.

For your information the slice preview will be full if you select (shift+arrow) a longer area which is adequate to the slice length.