polyend/TrackerBetaTesting #1392: [1.4.0] "Save" has not saved a project

rtannala, 2022-01-30:
Howdy,
I've encountered a one-off bug that hasn't happened to me before and I'm not currently able to reproduce it. I worked on a tune for a little while, not opening any other projects in between. I worked on the track in sessions and always saved before closing the device. After being worked on the project for a good while I decided to start a new one; saving the current one yet again before closing. When reopening that project later, I noticed that the project is in the state of the initial 1st save (i.e. when I've done "save as" and named the project). This means none of the times I've done just "save" have actually saved the project although I got the save progress bar each time. I checked from the SD card and the sample folder does not include the additional samples I've imported so the project seems to indeed be in the initial state and nothing has been written on any save.

Attached is the project file and this is what it was supposed to sound like: https://www.instagram.com/p/CYhawGJtCfX/

Bummer because I kinda liked the tune.

I'm on firmware 1.4.0, this has never happened before. I think I'll update before working on anything further.

[tracker_bugreport.zip](https://github.com/polyend/TrackerBetaTesting/files/7966807/tracker_bugreport.zip)

Cheers,
Riku

Jacknametrouble, 2022-01-31:
Hey Riku, thank you for your report. Might be connected with #1370. @MortalJS will try to replicate and confirm.

MortalJS, 2022-01-31:
Hi @rtannala 

Unfortunately, I was not able to reproduce this case. The last topic  #1370 was about a different case.

Jacknametrouble, 2022-02-01:
@rtannala 

Riku, devs attempted to rescue the project. It seems like the changes you've made were not saved for some reason. As soon as you've opened another project, the workspace folder which holds all the autosaving data was overwritten. The changes are lost. This is a single case of such behavior which is not easily replicable. The fact that it happened on 1.4 is not bringing us any closer to answers as some changes in the file handling engine was changed in 1.5 firmware. Therefore, we're closing this case. Thank you for your help, and apologies for the inconvenience.

closed 2022-02-01 by Jacknametrouble

rtannala, 2022-02-01:
@Jacknametrouble no worries, I was kind of expecting this assuming how the temp files would work. Seems like the project has stayed only in the workspace folder and the save function has for some reason or another failed. Luckily there's a newer firmware to move to, which would most likely at least reduce the risk of this happening again (unless it's a hardware glitch, which I doubt). Thanks for looking into it!

Jacknametrouble, 2022-02-02:
@rtannala Riku, hope this was a one-shot thing. But if anything like this would ever happen, there's a hidden workspace folder in the SD root dir. As long you didn't load another project. It should be holding all the changes. Having this, we have a chance to catch the glitch and maybe save the project from being lost. Best,