polyend/TrackerBetaTesting #1393: [1.5.0] Delay Settings are not always applied when loading a project

anemlife, 2022-01-30:
If you load a project, sometimes the delay keeps the settings that were applied to the last project (the values are displayed correct but the sound is not the right one). I have to move every values to make them work again.

* How often does this bug happen? often

* Does this bug happen in previous firmware versions? can't tell

* Does this bug happen in every project? it happens randomly as far as I've seen

Jacknametrouble, 2022-01-31:
Thank you for your feedback. Can you please provide a video showing the issue and the problematic project? I'm unable to get the same result as you're mentioning. These would be much of a help. Thank you!

anemlife, 2022-01-31:
Will try next time it happens

anemlife, 2022-01-31:
https://youtu.be/gL-CrJx4A40 here's a quick video. I switched from a project with a slow synced delay to this project where the delay should be playing unsynced at 11ms. You can clearly hear that the delay is not playing at the displayed values. When I turn the time knob on/off it works again as expected.

Jacknametrouble, 2022-02-07:
@MortalJS any luck reproducing this?