polyend/TrackerBetaTesting #1400: [Firmware version] Bug title

HeyheyMrG, 2022-02-03:
## Hi, sorry you had trouble!

### Steps to reproduce

Please describe the steps to reproduce the bug. Tell us what you noticed
happening and what you expected to happen instead.

### Attachments and links

Often the best way to show the problem is to record a short video,
attach the Tracker project you were working on, and provide links.

Pro-tip: You can drag and drop images and `.zip` files directly to the edit
box when writing your bug reports. Videos are too big as attachments, you
will have to use an external service for this and provide us links. Please
don't use services that delete files after short periods of time.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")

* Does this bug happen in previous firmware versions?

* Does this bug happen in every project?

HeyheyMrG, 2022-02-08:
Yesterday I deleted the instruments in this particular project one by one and each time I deleted one I checked to see if the problem still existed, which it did.  I then did a test sample recording of some radio noise in the same project and the problem had stopped.

UnorthodoxPark, 2022-02-15:
It also happens to me with certein samples, not all. I still have no clue why or how. Prolly the same bug.

Do you see a correct behavior of the timeline in this window? I mean, for me, sometimes the red cursor just stays frozen at 0 second, even when I hear the sample. Other times it shows perfectly advancing thru the timeline.

HeyheyMrG, 2022-02-15:
The red cursor doesnt move when the sound plays in this particular project, you can see it in the video I made, so possibly a similar issue?? Anyway I'll double check that all volumes are set correctly, I thought I'd done this but you never know.  I'll do some more testing - thx anyway.

HeyheyMrG, 2022-05-20:
Couldnt re create on any other projects, but still has the error if i re-load the old project.

closed 2022-05-20 by HeyheyMrG