polyend/TrackerBetaTesting #1401: "tune" for midi notes in performance mode?

Resident-X-D, 2022-02-04:
I have a suggestion:
Could it be possible, that in performance mode, when you apply the " tune" effect to a track (note up or down), that this could also affect tracks, which send out midi (to a synth or something)? 
Or implement the midi equivalent of that into the effects which can be used in performance mode?

bndofc, 2022-02-06:
what firmware are you on? i'm pretty sure this has been a thing since 1.4.0

Jacknametrouble, 2022-02-06:
https://github.com/polyend/TrackerBetaTesting/releases/tag/1.4.0

"Tune" FX in performance mode now also works for MIDI notes,

closed 2022-02-06 by Jacknametrouble

Resident-X-D, 2022-02-07:
Just found out the problem was the connected hardware devise. You can't "slide" rompler sounds while they're playing, just on the hit.