polyend/TrackerBetaTesting #1402: Notepad

elsummers, 2022-02-10:
Requesting a Note Pad be added perhaps in the song mode where one could add notes and song lyrics. 
Perhaps this would be just a generic notepad 
But maybe it could be a wordpad document where one could write a song or notes in the wordpad and then transfer it to the tracker ...

However, if I had to just summarize my request. I'm requesting just a basic notepad option where I could add notes ,

sebpmtl, 2022-02-10:
I use that function a lot in Renoise actually. Would be nice.

arphex, 2022-03-26:
> a 2 line note allowed per pattern and nothing more would be a game changer.

or no limitation at all - since this would take only a few bytes

DRxZxL, 2022-06-01:
That would be killer. Man id even settle for more keyboard characters and symbols etc. Simple things, forward dashes commas n fullstops etc 😅