polyend/TrackerBetaTesting #1406: [v1.5.0] pads always play MIDI 01 notes in the sample recorder and save dialog

james000, 2022-02-16:
I created a blank project to send midi notes (channel 10) to my drum machine to record using the sample recorder. After capturing my recording, I discovered that each of the letters plays a MIDI note when pressed and not on the MIDI channel selected. 

1. Create a new project
2. Press Pattern (note, still in Play mode)
3. Press Instrument - select 60. MIDI 10
4. Press Sample Recorder (source line in stereo, monitor on)
5. Press Record (on screen button)
6. Press a few keys to play some notes - Problem 1 - The tracker is now sending notes on MIDI Channel *1* instead of *10* as selected in step 3. 
7. Stop the recording (at this point the pads play the recorded sample correctly)
8. Press Save (on screen button)
9. Problem 2 - Any keys typed for the file name will also play MIDI notes on the channel seen in step 6. 

Alternate attempt with notes in the pattern to see if MIDI channel 10 sticks - it does not:

1. Create a new project
2. Press Pattern (note, still in Play mode)
3. Press Rec button
4. Press Instrument - select 60. MIDI 10
5. Press C6 to record a note in Track 1, Cell 1 using M10 (leave it in red mode)
6. Press Sample Recorder (source line in stereo, monitor on)
7. Press Record (on screen button)
8. Press a few keys to play some notes - Problem 1 - The tracker is now sending notes on MIDI channel *1* instead of *10* as selected in step 4. 
9. Press the Play hard key to start the pattern entered in step 5 and it will play on the correct MIDI channel.
10. Stop the recording (at this point the pads play the recorded sample correctly)
11. Press Save (on screen button)
12. Problem 2 - Any keys typed for the file name will also play MIDI notes on channel 1. 

Creating a project with *no* MIDI instruments or tracks still sends notes on MIDI channel 1 during sample recording and the recording save dialog. 

Thanks,
James

james000, 2022-02-16:
Confirmed similar behavior in v1.4.0

MortalJS, 2022-02-20:
@ziej8asek

closed 2022-03-19 by MortalJS

Jacknametrouble, 2022-04-29:
Fixed in https://github.com/polyend/TrackerBetaTesting/releases/tag/1.6.0b2