polyend/TrackerBetaTesting #1407: [1.5.0] .MOD files have empty instruments (no sound) when saved as project

Sebos1979, 2022-02-16:
## Hi, sorry you had trouble!

### Steps to reproduce

Import a .MOD
Play MOD : ok
Save it as a project
Open the newly created project
Play : no sounds
instruments are empty - no samples are loaded

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
Every time

* Does this bug happen in previous firmware versions?
No, working fine in previous firmware version

* Does this bug happen in every project?
Yes

Jacknametrouble, 2022-02-16:
Hey, thank you for your feedback. 

Note: The .mod file import and .it export may not 100% work as expected.

Accomplishing full unity in terms of exports and imports is not entirely achievable. Even when considering a standardized level of compatibility. It’s not possible to convert or provide data to convert projects one to one. There’s a variety of tracker applications out there, they’re working on a different basis usually.

And that’s also why the import/export capabilities are limited to the currently supported parameters and values. In other words, you may need to adjust the instruments and their particular values, maintain samples and volume levels manually.

Source; https://polyend.com/manuals/tracker/#loading-saving-projects

closed 2022-02-16 by Jacknametrouble

Sebos1979, 2022-02-16:
Hello Jacek,

Thank you for your answer.

I’m aware of the limitation due to the difference of architecture between trackers, and i understand there will be lost data if by example a Mod with 4 fx lane is imported in a tracker with 2 Fx lanes.

But the thing is that, since V1.5 and the introduction of the instruments, when i save the imported Mod as a PT project, almost none of the samples are kept in the conversion process to PT format. Steps and parameters are ok (with the known limitations) This was not with v1.4.

When i play within this process : Import Mod > Play mod  : it is ok
but when i play with this process : import Mod > save (or save as) then play from the PT project will lose mostly all the samples - the steps and some parameters are present, but instruments are empy.

I attached the stardstm.mod that was working before (Stardust Amiga Demo scene), you can see it has 16 working samples (aka instruments) and the converted project to PT format stardstm.zip (i had to zip the folder). you’ll see that steps and parameters like Glide and Volume are here, but there is only one instruments present.

I just wanted to point out a difference of behaviour beteween V1.5 and previous version.

Best regards,
Sébastien

Sebos1979, 2022-02-17:
Hello Jacek,

i’ve found the issue and the resolution.

The orginal Mod file had special characters on instruments and sample names. It seems like since the introduction of instruments in PT, it became more restrictive and instrument/sample with special characters are just ignored (not only the bad characters as in V1.4).

I modified the bad names in Milky tracker and now, my PT project from the mod is ok.

Best regards,
Sébastien