polyend/TrackerBetaTesting #1410: [v1.5.0] Reverb on instrument (volume -24.00 dB)

PITmast, 2022-02-21:
Hello everyone

There is an issue with reverb effect (with other instrument effects too probably).
1. load long sample (pad, drone, various texture);
2. set high reverb (the amount doesn't matter but 0.00 dB is good for test);
3. In Instrument Parameters 1/2 set Volume to -24.00 dB;
4. play note once on pads (beautiful reverb only without dry signal);
5. play note again - no sound!!!!!

To get it again: set Volume to 0.00 dB, play, set Volume to -24.00 dB and play.

Have a posibility to get wet signal only of single instrument would be a great feature, to be honest.
For instance, new value: {"wet", "-inf dB", "-24.00 dB",....}.

regards

Jacknametrouble, 2022-02-23:
Any chance we can get a video showing the mentioned issue? Thank you for your help.

PITmast, 2022-02-24:
Hi

Here is: https://mega.nz/file/z2AUkT5A#iCN8FAfiDtb87CVQQaPo5RCq4ylcVKqN1NLLSh0zzM4

Regards

closed 2023-04-26 by MiroPoly