polyend/TrackerBetaTesting #1411: Make 16 grid length option

laxmodin, 2022-02-21:
Hello polyend, 

A suggestion about UI.

I would like an option where grid 1 - 16 is visible at the same time, both in horizontal and vertical mode, when pattern top info is off. 

Right now we have pattern 1-15 showing in vertical mode, with half of the screen black at start. In horizontal we see grid 1-12. 

I think it would make for an easier understanding of the divisions of the bar if it followed the sizing of the screen. As in: half the screen equals half the bar, one quarter of the screen, one quarter of the bar. 

Especially useful when looping a pattern length of 16, as the screen never would jump around.

Jacknametrouble, 2022-02-21:
https://polyend.com/manuals/tracker/#general-midi-config
Pattern top info – Additional info, Track names (default), Off.

closed 2022-02-21 by Jacknametrouble

laxmodin, 2022-02-21:
That didnt help, jacek..