polyend/TrackerBetaTesting #1417: [1.5] Wrong steps when pasting in song mode

JohnAlchemy, 2022-02-22:
Hello,

If i copy a pattern, in song mode, which has less steps (24) when pasting, the new pattern will have 36 steps.

### Steps to reproduce

Reduce the steps of a pattern, copy paste it on song mode 

### Attachments and links  
N/A

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
It happens all the time 

* Does this bug happen in previous firmware versions?
I dont know if it happened on previous versions
* Does this bug happen in every project?
Havent try.

Thanks

Jacknametrouble, 2022-02-23:
Can we please get a video showing the issue?

JohnAlchemy, 2022-02-24:
Sure, [here you go](https://drive.google.com/file/d/1E0Wx2l1cILYee2jMFBFuDVv-jvaWUzma/view?usp=sharing)