polyend/TrackerBetaTesting #1418: [1.5.0] MIDI gate lengths problem with the Poly2

pawal, 2022-02-24:
## Hi!

### Steps to reproduce

I am using the tracker with the Poly2 (using the latest firmware). Make a track with a MIDI instrument with no other fx. Setting no MIDI length makes the Poly2 to miss gates. Setting the gate length in the track for each note fixes the problem. See below. This is _not_ using USB MIDI (have not tried it yet).

### Attachments and links

The first video shows a MIDI track with no gate length. The other half shows the Intellijel Quadra that misses the gates:
https://photos.app.goo.gl/3GPZUwoAy7iQdxBbA

The other one shows a MIDI track with gate length set to 8 (any gate length will do, anything between 1 and whatever):
https://photos.app.goo.gl/r584WmvrAGbskEoY8

So I would say that it is the Tracker that fails to send proper gates?

It is the only MIDI going on in this pattern.

### Other important information

* This happens all the time for me.

hypnogogue, 2022-03-22:
Hey - you need to set MIDI Off notes I think. Edit a step and scroll the jog wheel to the left until you see OFF. You’ll need to set an OFF event in between every MIDI Note. Or - setting gate length FX in essence sets the MIDI off note to fire at 0% to 100% of the step length, so you could also try setting Gate FX to 100 on every step.

pawal, 2022-03-28:
> Hey - you need to set MIDI Off notes I think. Edit a step and scroll the jog wheel to the left until you see OFF. You’ll need to set an OFF event in between every MIDI Note. Or - setting gate length [...]

Yes, if I set the note off between the notes, the 10% drop of notes does not happen. However, why is there even an inconsistency here? Why is there a note that gives me a trigger on if I never set a note off? I would be happier if that if I don't set a gate length, or a note off, there will not be a note on/off at all.

pawal, 2022-04-19:
If I never send an OFF, the next note should then not send a trig, right?

pawal, 2022-04-19:
> I would have thought you would just have a tied note (or slide if on a 303 for example)

Yes, that would have been the preferred alternative to this random triggering.

closed 2023-08-21 by borsukpolyend