polyend/TrackerBetaTesting #1420: 1.5.0

cooptrol, 2022-02-27:
This happens almost every time (see attached video) I switch it on, since I got it 3 weeks ago. After 1 or 2 reboots, it boots ok. Sometimes it boots with the correct graphics but the buttons are unresponsive. Since it still works fine after rebooting, I havent made any bigger complaint. I can't return it cos I live in Uruguay and it was delivered here by an intermediary courier. I guess it's not a software proble, or is it?

https://user-images.githubusercontent.com/100333241/155889475-21df16dc-a187-4492-83a4-67fcee2755ea.mp4

Jacknametrouble, 2022-02-28:
If you're using the original AC adapter and cable, try using an alternative one, maybe the original AC adapter is causing issues.

Jacknametrouble, 2022-03-01:
Does this mean, you solved the issue?

cooptrol, 2022-03-02:
Just tested a different charger and first boot was ok, gonna keep reporting in the next days. Hope this is the solution!

cooptrol, 2022-03-04:
Seems that was the problem! A shame not to be able to use the original charger, but it works fine with my phone!

bufu9k, 2022-03-06:
Hi, I’d just like to add that I am able to reproduce this bug very consistently by not shutting the device down properly. If I simply shut off the surge protector the tracker is plugged into, on next boot it will do the wonky graphics. 

Sometimes it takes two or three tries afterwards but I’m always able to get it back to a proper boot and into a stable state. 

This happened the very first time I powered up the device and I was FLOORED because I thought I got a defective unit. Nope, just something weird. Can’t tell if it’s hardware or software. 

v1.5.0. I’ve also tried the original power adapter, an official apple adapter and I tried via directly plugging into a wall socket or through a surge protector and the same result happens.

cooptrol, 2022-03-06:
With the different charger, the weird graphics are gone but sometimes the buttos are frozen when booting up.

closed 2022-03-07 by Jacknametrouble

cooptrol, 2022-03-08:
Why was this closed? The same bug I described using the original charger started happening with a different charger. This goes on. First 2 days worked fine, the again the same. Tested with a 3rd charger, same problem. What else can I do?