polyend/TrackerBetaTesting #1422: Playback not working. v1.5 and v1.3

myfriendscallmeJackie, 2022-03-03:
Hello-

My Play button functions but only the first line of any given pattern is played. Same in song mode. The clock at the top continues to run, but the screen does not scroll. Often the first line of any given selection is the only thing able to be previewed, often not able to be previewed at all. Playback in the sample editor does not function.

My up and down arrow keys only function in rec mode or while navigating menus.

Obviously, this renders the product useless so any advice helps.

(I've hard rebooted, changed firmware)

0x0zzz, 2022-03-04:
Have you changed the MIDI clock-in setting?  This sounds like the behavior when you press play while it's not master (clock-in set to USB or MIDI).

myfriendscallmeJackie, 2022-03-04:
You are the best!

And I feel like an idiot... ha. 

Thank you very much.

closed 2022-03-04 by Jacknametrouble