polyend/TrackerBetaTesting #1425: [1.5.0] Bring back the sample folder in projects

anemlife, 2022-03-13:
The sample folder has been replaced by the instruments folder which contains the .pti files.

I used to edit samples with the tracker and load them into ableton projects but I can't do it anymore as there is no sample folder anymore containing the originals .wav. 

Could you bring the folder back? Thanks for your help

Jacknametrouble, 2022-03-14:
No, we can't. Use the original samples from your SD card.

closed 2022-03-14 by Jacknametrouble

anemlife, 2022-03-14:
@Jacknametrouble 
That means we can't access edited .wav samples from a project without exporting them one by one or is there a way? Rollback is not an option. 1.5 fixes way too much bugs to rollback. 
Also: if I try to open a project saved with 1.5.0 on a previous version it will probably cause issues due to the replacement of the sample folder by the instrument one or am I missing something?

> " It's one way or another. " What about an option in the menu? (saving instruments as .pti on/off)

Jacknametrouble, 2022-03-18:
This means you can have access to the edited sample as an instrument, from the level of a project. As they say, you can't always get what you want. It's one way or another.

anemlife, 2022-03-18:
That's fine but it doesn't answer my question at all: can I roll back to 1.4 and get the .wav from my instruments back or are they lost now that I saved them with the new firmware?

Jacknametrouble, 2022-03-19:
If they're saved with the 1.5 then they're converted already.