polyend/TrackerBetaTesting #1426: [1.5.0] Rendering with external clock causes the tracker to freeze

anemlife, 2022-03-14:
If you try rendering a selection with clock source set to external the tracker freezes. Render option should not be available with external clock (at least greyed out) so users can't accidentally make the machine crash.

Synthdawg, 2022-03-23:
I think im seeing same. Im on 1.6. Rendering stems is OK. Then changing MIDI to external 'Clock in' i.e. USB or MIDI Jack in, shows a frozen progress bar when attempting to render stems. I know that it doesn't export stems with external clock but I would have expected a message when attempting this. Also when reverting to the internal clock then it won't render at all until powered off / on to reset this.

arphex, 2022-04-07:
I also confirm the same - render selection with clock in/transport in hangs the tracker and restart is needed.   I think graying out the option is not really a solution - just fix the bug so that code waits for clock in/transport in and then renders, or just switches temporarily to internal clock.  

Making it non usable when tracker uses clock in makes composing with the tracker really difficult as  this is primary way to work with limitations

closed 2022-04-29 by Jacknametrouble