polyend/TrackerBetaTesting #1428: [1.5.0] Export as .it does nothing. Remove menu item if discontinued

bambrose, 2022-03-18:
### 

1. Create Songe
2. File / export as .it
3. Watch progress bar
4. Open SD card on computer. No .IT files (AT ALL) on the card (IN ANY FOLDER)

### Attachments and links
Nothing to attach

### Other important information

I carried out a 'song expot' (aka render) beforehand.. not sure if that somehow jinxes it ? or whether the function has simply been retired, but the option not yet removed from the menu ?

Jacknametrouble, 2022-03-18:
Can't confirm, works ok here. Please provide a video showing the issue. Please be nice with language, we're kinda ok with commands on the company level already boss.

bambrose, 2022-03-20:
Sorry JackNameTrouble (Jacek ?)
My error. I was looking in export and not in MOD folder.
Found it now, apologies.
I genuinely thought the functionality had been retired. My mistake, apologies for any offence caused.
.. Only found the file after recording a video for you.. so it's a pretty good strategy to get people to do that.. an error check in itself,

Peace,
Bill

closed 2022-04-29 by Jacknametrouble