polyend/TrackerBetaTesting #1429: Make font change optional

keith-macleod, 2022-03-18:
Hi, Just checking out 1.6.0b1, and I know you probably don't want to hear this, but I like the old font better. It may not be the prettiest but I find it easier to read, and that's really the more important thing as far as I'm concerned. It would be nice to still maintain the option to use the old font instead.

I can totally understand if font management is something very low down on the list of things you want to deal with.

Thanks for all the great work as always.

trutypesounds, 2022-03-19:
> Hi, Just checking out 1.6.0b1, and I know you probably don't want to hear this, but I like the old font better. It may not be the prettiest but I find it easier to read, and that's really the more [...]

i guess the readability problem not in the font itself, but the font kern (kerning) - the space between symbols.
it looks odd in places and completely broken in specific screens (renaming, naming sections)
see the screens to get the idea
[image]
[image]
[image]
[image]

keith-macleod, 2022-03-20:
There's that, too. I still much prefer the orginal font, kerning issues aside.

bambrose, 2022-03-20:
Me too I’m afraid . I appreciate the improvements and they do look good and much neater. But the font looks “fatter” and is as a consequence less legible. I’ve reverted to 1.5.0 for the time being for that reason.
It’s a shame. I can appreciate that someone has spent a lot of time getting it to look right.. it’s a very subjective thing

MetabaronCZ, 2022-03-20:
How about monospace fonts used in programming IDE (Consolas for example)?

Synthdawg, 2022-03-22:
+1 for optional fonts. If its a choice I like the original best.

bignutemporium1, 2022-03-24:
Just wait for the inevitable homebrew firmware;

https://www.reddit.com/r/PolyendTracker/comments/tem857/retracker_reverse_engineering_framework_for_the/

https://github.com/patois/RETracker/issues/2

>I have a copy of Polyend's internal source for the Tracker (and their next device if you read the loads of documentation they include, NEARLY ENTIRELY IN ENGLISH, with only short release/change-notes in Polish.

This is includes the extra files used to run this software off just open-source technology like Teensy 1.6 and some Arduino._ and will certainly make reverse engineering it easier.

These files were not acquired by malicious means and their licensed for distribution (!!!)

[SourceCode.zip](https://github.com/patois/RETracker/files/8254049/SourceCode.zip)
provided by JerzyHertz/EgoShiner/Aaron

theorize999, 2022-03-25:
I wish I had time to design a font I've always wanted to do that... what format does PT use? 
I've been waiting for the homebrew firmware to show up..;. I knew it was only a matter of time. I thought that the firmware was open source.... I'm wrong about that?

0x7CA, 2022-03-25:
> I wish I had time to design a font I've always wanted to do that... what format does PT use? I've been waiting for the homebrew firmware to show up..;. I knew it was only a matter of time. I thought that the firmware was open source.... I'm wrong about that?

It's not open source, this repo is just to track releases and to provide feature requests and bug reports.
Regarding the font, you might be able to patch the bitmap font with your own font. 
Use `fnt_cvt` https://www.ftdichip.com/old2020/Support/Utilities.htm#EVEFontConverter to get the raw data then patch it over the current font data with RETracker

jonwaterschoot, 2022-04-01:
[image]
I also experienced the new font unfit, as the **faulty kerning** is a big issue.
I'm sure there are people better at pointing out the reasons and solutions. Font rendering can be complicated. 
Also: I see the potential for using a smaller font as you do win space, but I felt I had to get closer to the device, readability is of greater importance. Yet, I'm sure there will probably be a middle ground to be found.
(I really do love the new delay options however.)

plcktzh, 2022-04-11:
May I suggest IBM Plex Sans as a compromise between the old font and the new Helvetica-ish one? ;)

MetabaronCZ, 2022-04-13:
I think it is not necessary to use high resolution antialiased fonts. They could use some monospace pixel font, render app in quarter resolution and then scale x4 to full display resolution (like game emulators do).
This could save some resources for sound engine, but the UI would look similar to Digitakt :).