polyend/TrackerBetaTesting #1434: [1.5.0] MIDI Gate FX + Micro move/Roll = ignored Gate FX

hypnogogue, 2022-03-22:
### Steps to reproduce

1. Record a MIDI note
2. Add Gate FX in a FX slot > set to 50 (or any value)
3. Add Micro-move in other FX slot > set to any value 

Behavior: Gate FX is ignored (midi note off does not fire at end of gate)

Expected: Gate FX starts note at Micro-move time, ends note/sends MIDI off honoring the percentage of the step time (ex: step time is 100 ms. Gate FX is 50%. Micro-move at 75. Note starts at 75ms, ends at 25ms of next step. If note occupies next step, send MIDI off at start of note - like a monophonic synth would) 

* How often does this bug happen? Every time

* Does this bug happen in previous firmware versions? Unsure

* Does this bug happen in every project? Yes

hypnogogue, 2022-03-22:
Similar issue when Roll + Gate FX are on a MIDI note step

doelman, 2022-03-22:
addition: this doesn't only apply to MIDI notes. Gate length FX is also ignored for samples / instruments when Micro-move is also on the same step.