polyend/TrackerBetaTesting #1440: Omnichord mode for Midi chord FX

reubenfinger, 2022-03-26:
The midi chord fx laid out across the 48 tracker pads like an omnichord.

each row selects a chord type of the twelve notes in the scale, eg

Row 1 chooses which octave.

Row 2 selects a major chord for each of the twelve notes in the scale.

Row 3 selects a minor chord.

Row 4 selects a seventh chord.

Ideally these could be played and recorded live into the sequencer and out to midi gear.

The tracker is amazing by the way thank you all for your constant hard work.

bambrose, 2022-03-26:
I mean this would be a coup if it could be done.
That degree of sophistication on the chord side of things would really open it up

ruscular3d, 2022-05-08:
If we could program which chord we want on each row would be awesome!