polyend/TrackerBetaTesting #1441: [1.5.0] Can't enter directory starting with special character

0x7CA, 2022-03-26:
### Steps to reproduce

In sample loader list a directory that starts with `$`, then try to enter the directory. It will go to the parent directory instead of entering the directory.
Does not have exactly the same behaviour as entering `/..`, the cursor will stay on the directory that it traversed from, while `/..` resets the cursor to the top.

I tried it with the characters `$` and `&`, but I assume it's all special characters that get indexed before `/..`

[image]

0x7CA, 2022-03-26:
Also present in 1.6.0beta1

Jacknametrouble, 2022-03-26:
Thank you for your feedback. Special characters are not supported by the Tracker.

closed 2022-03-26 by Jacknametrouble