polyend/TrackerBetaTesting #1444: [1.5 & 1.6b] Clicky, stuttery, static like effect on sample playback with antialiasing off

trutypesounds, 2022-03-28:
Well, when antialiasing is off samples sound kinda clicky through, stuttery, like radio static or something and im talking here about root note playback, so it's not that ringy aliasing sound when you play different notes, its different. 
At first i thought it was related to samples sampled off radio, but yesterday i've sampled some mellow fm tones from Elektron Cycles and i was in headphones and mentioned that unpleasant effect on samples.
So im not sure if its a bug, or bad sample playback algo implemented, but i think that must be improved.

p.s. i guess that was present in previous firmwares also, 1.4 too probably, cant test it, but def it was in 1.5 and still in 1.6

Jacknametrouble, 2022-03-28:
Thank you for your feedback. Can we please get a video showing the issue? Would be very helpful.

trutypesounds, 2022-03-30:
> Thank you for your feedback. Can we please get a video showing the issue? Would be very helpful.

Here are videos with samples off Elektron Model Cycles, one is mellow and the second is a noisy metallic percussion sound, please use the headphones so you hear clicky clicks in mellow chord and a prominent click in metallic perc sound when antitaliasing is off.

https://user-images.githubusercontent.com/17484006/160917610-b5f3e49c-ed23-4864-85fd-3164c10c7dd9.MP4

https://user-images.githubusercontent.com/17484006/160917665-d2e5fec8-c004-4f07-93df-7978caa351cc.MOV

bndofc, 2022-04-30:
i can confirm this, when the root note of the sample is played with anti-aliasing off there are some pretty weird artifacts, only way of avoiding this is by using fine tuning or not using sample root note :(