polyend/TrackerBetaTesting #1445: [1.6.0b1] Sample preview on sampling screen still kinda buggy

trutypesounds, 2022-03-28:
Well, its better in 1.6 but still buggy behavior - when sound is sampled, pressing two notes at once with a slight delay causes the latest note pressed to sound louder than previous one.

Jacknametrouble, 2022-03-28:
Thank you for your feedback. Can we please get a video showing the issue? Would be very helpful.

trutypesounds, 2022-03-29:
will try to film this, meanwhile you can try to replicate the issue

trutypesounds, 2022-03-30:
> Thank you for your feedback. Can we please get a video showing the issue? Would be very helpful.

well, i've tried to replicate that with no success, so lets close this one i guess for now.

trutypesounds, 2022-04-02:
@Jacknametrouble well, its still here, dont know how to replicate this at will, but the bug appeared while i was working with tracker for a couple of hours, suddenly the sample preview after sampling was quite very loud (this glitch was more prominent on previous firmwares and was very annoying for headphone users).
check the first video, after sampling the preview is very very loud (its painful in headphones)

https://user-images.githubusercontent.com/17484006/161398990-0288905f-330b-4e29-b171-35391c741c7f.MOV

on the second video i've rebooted the tracker and sample preview after sampling is fine

https://user-images.githubusercontent.com/17484006/161399013-3cad71e7-cf3f-4bdf-a9fa-827a3061b539.MOV

closed 2024-02-14 by MiroPoly