polyend/TrackerBetaTesting #1446: [1.6.0b1] Reverb predelay barely noticable

trutypesounds, 2022-03-28:
Is it me, or reverb predelay barely noticeable (or not doing anything at all) on some sounds? Or is it that very subtle?

Jacknametrouble, 2022-03-28:
Thank you for your feedback. Can we please get a video showing the issue? Would be very helpful.

trutypesounds, 2022-03-29:
will try to film this

trutypesounds, 2022-03-30:
> Thank you for your feedback. Can we please get a video showing the issue? Would be very helpful.

here is a couple of videos, it's working, but sometimes its very very subtle, so it seems like there is nothing going on when changing predelay parameter. 
may be im used to predelays being prominent.

https://user-images.githubusercontent.com/17484006/160915450-c0952ac2-4c08-4344-aacd-12a808cfe0fd.MOV

https://user-images.githubusercontent.com/17484006/160915497-670c5c88-2f23-4b6c-b53c-4835c8cf0a94.MOV