polyend/TrackerBetaTesting #1451: [1.5.0] Sample Loader Issues

JonathanTaquet, 2022-04-02:
## Sample Loader issues when "Preview" button is pressed

Hi, I have found some issues in the `Sample Loader` page when performing things while the "Preview" button is pressed.
These bugs are related to the issue #505 I had previously opened, but that is now closed.
I will try to be more exhaustive here, in order to try letting you better understand the full problem.

All these glitches may be linked to a common software issue, so I group them all together here.
I will also mention some similar cases which actually work, to provide expected behavior.

**Notes**:
- It also appears that after encountering some of the glitches, under some conditions, the same **glitches will also happen even if the "Preview" button is not pressed** (see details below).
- In at least one case, the issue becomes a major issue: **the Tracker seems to become completely stuck** and you could need to unplug power to stop/restart it.

### Steps to reproduce

By following the steps bellow, in the right order, you will encounter all the mentioned issues **Issue $x**, for x in the range of 1 to 5.

- Start with a blank project:
  - press "File" button
  - press `New Project`
- Go to the `Sample Loader` page by pressing "Sample Loader" button.
- In the `SD card` panel, select a folder containing samples, and select a sample.
- Press `Preview` button to listen to preview of the sample on the SD Card, and while holding `Preview` button, press `Add`, `Add next` or `Import` button.
**Issue $1**: nothing happens where I would have expected my sample being added to the instruments list.
**Possible fixes**:
  - either grey out the `Add`, `Add next` and `Import` buttons when `Preview` is pressed to indicate that adding/importing is not feasible when previewing,
  - or effectively add/import the sample properly. This option would be preferred and should be feasible as it is already working properly when listening to preview of the samples in the `Instruments` list/panel, while pressing `Add`, `Add next` or `Import` to add/import samples (see **Notes** below).

- Release `Preview` button, and press `Add` to effectively add the sample to the empty slot that is currently selected in the `Instruments` list.
- Press `Preview` again (and eventually select another sample) to preview sample(s) on the SD Card and select a sample.
- While holding `Preview` button, press `Import` button.
  - A popup appears with the message `This action will replace the instrument. Do you want to continue?` and the buttons' menu is changed to display `No` and `Yes` over the two last buttons.
  - Now, **release the `Preview` button** (this is to prevent inadvertently obtaining **Issue $4** below)
    - **Note**: if your previewed sample is long enough, you will hear that the playback of the sample will however continue until it reaches the end this may be seen as an additional issue, but it is not especially problematic.
  - if you press `Yes`, **Issue $2** happen:
    - the popup disappear, but the import window does not appear and nothing happen.
    - the button's menu is still showing `No` and `Yes` but if you press anyone of the two buttons, nothing happens.
    - but, you can navigates with the arrows between the samples and between the panels, and normal buttons behaviour works as if the correct button's menu was shown.
    - to restore correct button's menu, you can either:
      - press the button normally at "Add" position (1st one from the left), and answer `No` to the popup,
      - or press the button normally at "Add next" position (2nd one from the left), and answer `No` to the popup,
      - or press the button normally at "Import" position (5th one from the left), and answer `No` to the popup,
      - or navigate to the `Instruments` panel and either:
        - do one of the 3 previous solutions,
        - or press the button normally at "Rename" position (4th one from the left)
          - The sample renaming window will appear,
          - just close it by either pressing `Cancel` or `Rename`,
        - or press the button normally at "Delete unused" position (6th from the left)
          - A popup appears with the message `This action will delete all unused instruments. Do you want to continue?`
          - Answer either `Yes` or `No`, if you have released the `Preview` button when indicated above, else answer `No`.
    - now the menu is restored. 
  - **Note** if you had pressed `No`, to skip **Issue $2** the popup would have disappeared properly and nothing would have been done, as expected.
- if you have not released the `Preview` button when indicated above, you may need to restore a << normal >> state of the Tracker (see **Issue $4** below).
  - navigate to a sample in `SD Card` panel and press then release the `Preview` button at least one time.
    - **Note**: it may happen that the first time you press `Preview`, the preview of the sample is not listened (see **Issue $5** below), but even if nothing is listened, the normal state would be restored.

- Select a sample in the `SD Card` panel and press `Preview` button.
- Press `Add` or `Add next` while holding `Preview` button.
  - A popup appears with the message `This action will replace the instrument. Do you want to continue?` and the buttons' menu is changed to display `No` and `Yes` over the two last buttons.
  - Now you can release or not the `Preview` button, as you wish, it will works the same (you already entered into an abnormal state, see **Issue $4** below).
    - **Note**: if you release it and your previewed sample is long enough, you will hear that the playback of the sample will however continue until it reaches the end this may be seen as an additional issue, but it is not especially problematic.
  - if you press `Yes`, **Issue $3** happen:
    - the popup stay visible, the buttons' menu disappear and nothing happens... The Tracker seems to be stuck.
    - to unfreeze the unit you need to press on the button previously corresponding to the answer `No` (the previous last one)
  - **Note** if you had pressed `No`, to skip **Issue $3** the popup would have disappeared properly and nothing would have been done, as expected, but you would have still entered in an unexpected state of the Tracker (see **Issue $4** below).

From the time `Preview` is hold to preview a sample from the `SD Card` panel and at the same time you have pressed `Add` or `Add next`, the Tracker has entered into an unexpected state, and it will occur every time you do that... (until you press and release `Preview` to preview a sample of the `SD Card` panel).
- For the moment, do not press `Preview` anymore
- Select a sample in the `SD Card` panel
- Try to use `Add` or `Add next`
[... 30 lines cut ...]
- **Issue 1** will happen even if there are already samples in the instrument list, when you are trying to add/import to an empty slot while previewing samples in the `SD Card` panel.
- **Issue 2** and **Issue 3** will happen when trying to add/import samples to a slot in the instruments list that is not empty, while previewing samples in the SD Card.

If the tracker is not inside of the unexpected state mentioned above, there is no issue when previewing samples in the `Instruments` panel while performing other actions:
- Pressing "Backspace Delete" button while holding `Preview` button to delete a sample being previewed in the `Instruments` panel works properly.
- Pressing "Add", "Add next" or "Import" button while holding `Preview` button to replace a sample being previewed in the `Instruments` panel works properly.

### Attachments and links

I don't think attachments are needed, you should be able to reproduce that easily.
But if you really need something for particular cases you are not able to reproduce, please ask...

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")

These issues happen every time, you should be able to reproduce them easily.

* Does this bug happen in previous firmware versions?

Yes, Similar bugs happened in previous firmware versions. Maybe not the **Issue $1**, I don't remember, but **Issue $3** for sure.

* Does this bug happen in every project?

I would say yes, but if I remember, when loading a project coming from a previous firmware version, the behavior looked quite different. But similar issues to reported ones were obtained when using that previous firmware version with the same project.

Jacknametrouble, 2022-04-29:
Fixed in https://github.com/polyend/TrackerBetaTesting/releases/tag/1.6.0b2

closed 2022-04-29 by Jacknametrouble

JonathanTaquet, 2022-07-19:
It looks far more usable now (v1.6.0), thanks for the great improvement